rust-embedded/riscv #11: Raw instruction functions are unsafe footguns

strake, 2018-08-18:
Except `wfi`, they are effectively impossible to use safely: Rust can move values arbitrarily between registers (or memory!) behind the scenes, so conforming to an ABI is impossible; to do so, one would need to specify the argument and return registers in the `asm` block itself.

I vote to merely delete these functions (again, except `wfi`).

See also https://github.com/rust-embedded/riscv/pull/10

danc86, 2018-08-19:
Sorry, I don't understand what you mean here. Take `riscv::register::mepc::read()` for example. The inline asm block tells LLVM that it needs to allocate a register for the value to be read into. I'm not sure what you mean "Rust can move values arbitrarily behind the scenes"... Certainly whatever happens *around* the inline asm block is Rust's/LLVM's business and it can and will emit code to move values around between registers. But it will still schedule a suitable register to fill into the inline asm block.

To ask another way, how can calling `riscv::register::mepc::read()` lead to unsafe behaviour at runtime?

Certainly, *changing* the CSR values needs to be done with care... but I think all the functions which set or clear CSR bits are already marked unsafe, aren't they?

dvc94ch, 2018-08-19:
Would also be interested to know. I quoted the spec below to help with discussion

> Write Any Values, Reads Legal Values (WARL)
Some read/write CSR fields are only defined for a subset of bit encodings, but allow any value to be
written while guaranteeing to return a legal value whenever read. Assuming that writing the CSR14
Volume II: RISC-V Privileged Architectures V1.10
has no other side effects, the range of supported values can be determined by attempting to write
a desired setting then reading to see if the value was retained. These fields are labeled WARL in
the register descriptions.
Implementations will not raise an exception on writes of unsupported values to an WARL field.
Implementations must always deterministically return the same legal value after a given illegal
value is written.

> 3.1.19
Machine Exception Program Counter (mepc)
mepc is an XLEN-bit read/write register formatted as shown in Figure 3.20. The low bit of mepc
(mepc[0]) is always zero. On implementations that do not support instruction-set extensions with
16-bit instruction alignment, the two low bits (mepc[1:0]) are always zero.
The mepc register can never hold a PC value that would cause an instruction-address-misaligned
exception.
mepc is a WARL register that must be able to hold all valid physical and virtual addresses. It
need not be capable of holding all possible invalid addresses. Implementations may convert some
invalid address patterns into other invalid addresses prior to writing them to mepc.
When a trap is taken into M-mode, mepc is written with the virtual address of the instruction that
encountered the exception. Otherwise, mepc is never written by the implementation, though it may
be explicitly written by software.

dvc94ch, 2018-08-19:
@danc86 @strake is talking about something else xD
https://github.com/rust-embedded/riscv/blob/master/src/asm.rs#L21-L25

> 3.2.1
Environment Call and Breakpoint
The ECALL instruction is used to make a request to the supporting execution environment.
When executed in U-mode, S-mode, or M-mode, it generates an environment-call-from-U-mode
exception, environment-call-from-S-mode exception, or environment-call-from-M-mode exception,
respectively, and performs no other operation.
ECALL generates a different exception for each originating privilege mode so that environment
call exceptions can be selectively delegated. A typical use case for Unix-like operating systems is
to delegate to S-mode the environment-call-from-U-mode exception but not the others.
The EBREAK instruction is used by debuggers to cause control to be transferred back to a debug-
ging environment. It generates a breakpoint exception and performs no other operation.
As described in the “C” Standard Extension for Compressed Instructions in Volume I of this
manual, the C.EBREAK instruction performs the same operation as the EBREAK instruction.
ECALL and EBREAK cause the receiving privilege mode’s epc register to be set to the address of
the ECALL or EBREAK instruction itself, not the address of the following instruction.

There must be a way to pass arguments through registers with `ecall`. So the ecall instruction is not useful at the moment if the syscall requires arguments.

danc86, 2018-08-19:
Oh! Okay I get it now. Yeah, the `ecall` and `ret` ones can't really be used as is.

`ebreak` doesn't have any register semantics, it's certainly useful as is, but yeah not sure if it really meets Rust's definition of "safe". Marking all those functions as unsafe makes sense to me.

Sorry @strake I should have looked properly at the PR :man_facepalming:

strake, 2018-08-19:
@danc86 It was a failure to specify on my part 😆

As for `wfi`, i believe it ought to ultimately be called `core::sync::atomic::spin_loop_hint`.

dvc94ch, 2018-12-18:
Fixed in #14

commit a295a730 (2024-01-02): drop write macro in favor of IO objects + fmt::Write

closes #4
closes #5
closes #6
closes #7
closes #11
closed 2018-12-18 by dvc94ch