rust-embedded/riscv #27: rust cross compile to riscv64gc

H-Y-B, 2019-04-19:
I want help, please.
(0)This is my rustc:
rustc 1.34.0 (91856ed52 2019-04-10)

(1) I have installed riscv64gc-unknown-none-elf.
`hyb:~/Rust/riscv64$ rustup show
Default host: x86_64-unknown-linux-gnu

installed targets for active toolchain

armv7-unknown-linux-gnueabihf
**riscv64gc-unknown-none-elf**
x86_64-unknown-linux-gnu

active toolchain

stable-x86_64-unknown-linux-gnu (default)
rustc 1.34.0 (91856ed52 2019-04-10)
`
(2) This is my ~/.cargo/config

`[target.riscv64gc-unknown-none-elf]
linker="riscv64-unknown-elf-gcc"
`

(3) Then, i run this command
`cargo build --target=riscv64gc-unknown-none-elf`

(4)But it shows this:
`Compiling riscv64 v0.1.0 (/home/hyb/Rust/riscv64)
error[E0463]: can't find crate for `std`
  |
  = note: the `riscv64gc-unknown-none-elf` target may not be installed

error: aborting due to previous error

For more information about this error, try `rustc --explain E0463`.
error: Could not compile `riscv64`.

To learn more, run the command again with --verbose.
`
I want to know why?   linker or std or rustc version?

Disasm, 2019-04-19:
(2) You don't need this since Rust supports linking with lld out of the box.
(4) In embedded world applications are `no_std`. Look how it's done [here](https://github.com/riscv-rust/k210-example/blob/33deb8749f31ad171ef9fab80a52cdaa460c8941/src/main.rs#L2-L3).

What hardware are you targeting?

Disasm, 2019-06-03:
Closed due to inactivity.

closed 2019-06-03 by Disasm