rust-embedded/riscv #54: Set Stack Pointer and Return Address

dkhayes117, 2020-09-21:
I don't see anything in this crate for setting the stack pointer(x2) or return address(x1).  This is required to enter into a lower privilege level (user mode) within a defined memory stack.  The assembly should be a simple move instruction.  Finally an MRET instruction would be called to enter user mode.  I noticed that the MRET instruction was available in this crate in v0.2.0.  Is there another way to call it now?  Why was it removed?

dkhayes117, 2020-09-21:
Hmm..... is sfence_vma used to run these type of instructions?

Disasm, 2020-09-21:
I think that `MRET` was removed because it's useless on its own, you usually need to execute a few more instructions before it.

dkhayes117, 2020-09-21:
My pseudo code is 

Set MEPC with entry point of user level function
Set sp to the stack pointer of my user level stack (ra should be set to 0)
set MPP  to 00 for user mode
then MRET

Do I need inline assembly to do this in my program?  Isn't inline unstable currently?

Disasm, 2020-09-21:
You can use either inline assembly or precompiled binary hack (also used in `riscv` and `riscv-rt`) for this. We use the latter because it allows to use assembly on stable.

dkhayes117, 2020-09-21:
I would love to use precompiled binary for it, but I think that is beyond my skills.  I guess inline and nightly it is.

closed 2020-09-23 by dkhayes117