rust-embedded/riscv #59: linking with `rust-lld` failed: no memory region specified for section '.eh_frame', on Ubuntu 16 i686

advancedwebdeveloper, 2020-11-10:
I was trying to build some demo projects, by following [this](https://pramode.in/2019/10/07/rust-on-riscv-board-sipeed-longan-nano/) article.
Got these bugs:
> $ rustup target add riscv32imac-unknown-none-elf
> info: component 'rust-std' for target 'riscv32imac-unknown-none-elf' is up to date
> ~/Desktop/rust-sipeed-longan-nano-master$ rustup default nightly
> info: using existing install for 'nightly-i686-unknown-linux-gnu'
> info: default toolchain set to 'nightly-i686-unknown-linux-gnu'
> 
>   nightly-i686-unknown-linux-gnu unchanged - rustc 1.49.0-nightly (1773f60ea 2020-11-08)
> 
> ~/Desktop/rust-sipeed-longan-nano-master$ rustup target add riscv32imac-unknown-none-elf
> info: component 'rust-std' for target 'riscv32imac-unknown-none-elf' is up to date
> ~/Desktop/rust-sipeed-longan-nano-master$ rustup target add riscv32imac-unknown-none-elf
> info: component 'rust-std' for target 'riscv32imac-unknown-none-elf' is up to date
> oceanfish81@gollvm:~/Desktop/rust-sipeed-longan-nano-master$ cargo --verbose build
>    Compiling r0 v0.2.2
>      Running `rustc --crate-name r0 /home/oceanfish81/.cargo/registry/src/github.com-1285ae84e5963aae/r0-0.2.2/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=c85b5ab54aceb5f9 -C extra-filename=-c85b5ab54aceb5f9 --out-dir /home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps --target riscv32imac-unknown-none-elf -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/debug/deps --cap-lints allow -C link-arg=-Tlink.x`
>    Compiling panic-abort v0.3.1
>      Running `rustc --crate-name panic_abort /home/oceanfish81/.cargo/registry/src/github.com-1285ae84e5963aae/panic-abort-0.3.1/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=37322ac1e75fee7c -C extra-filename=-37322ac1e75fee7c --out-dir /home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps --target riscv32imac-unknown-none-elf -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/debug/deps --cap-lints allow -C link-arg=-Tlink.x`
>    Compiling rust-sipeed-longan-nano v0.1.0 (/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master)
>      Running `rustc --crate-name rust_sipeed_longan_nano --edition=2018 src/main.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 -C metadata=2df4b3c694b8dfb5 -C extra-filename=-2df4b3c694b8dfb5 --out-dir /home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps --target riscv32imac-unknown-none-elf -C incremental=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/incremental -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/debug/deps --extern panic_abort=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/libpanic_abort-37322ac1e75fee7c.rlib --extern r0=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/libr0-c85b5ab54aceb5f9.rlib -C link-arg=-Tlink.x`
> warning: unnecessary parentheses around assigned value
>  --> src/main.rs:8:25
>   |
> 8 | const rcu_apb2en: u32 = (0x4002_1000 + 0x18);
>   |                         ^^^^^^^^^^^^^^^^^^^^ help: remove these parentheses
>   |
>   = note: `#[warn(unused_parens)]` on by default
> 
> warning: unnecessary parentheses around assigned value
>   --> src/main.rs:10:25
>    |
> 10 | const gpioa_ctl0: u32 = (0x4001_0800 + 0x0);
>    |                         ^^^^^^^^^^^^^^^^^^^ help: remove these parentheses
> 
> warning: unnecessary parentheses around assigned value
>   --> src/main.rs:11:25
>    |
> 11 | const gpioa_data: u32 = (0x4001_0800 + 0xc);
>    |                         ^^^^^^^^^^^^^^^^^^^ help: remove these parentheses
> 
> warning: unused import: `panic_abort`
>   --> src/main.rs:13:5
>    |
> 13 | use panic_abort;
>    |     ^^^^^^^^^^^
>    |
>    = note: `#[warn(unused_imports)]` on by default
> 
> warning: constant `rcu_apb2en` should have an upper case name
>  --> src/main.rs:8:7
>   |
> 8 | const rcu_apb2en: u32 = (0x4002_1000 + 0x18);
>   |       ^^^^^^^^^^ help: convert the identifier to upper case: `RCU_APB2EN`
>   |
>   = note: `#[warn(non_upper_case_globals)]` on by default
> 
> warning: constant `gpioa_ctl0` should have an upper case name
>   --> src/main.rs:10:7
>    |
> 10 | const gpioa_ctl0: u32 = (0x4001_0800 + 0x0);
>    |       ^^^^^^^^^^ help: convert the identifier to upper case: `GPIOA_CTL0`
> 
> warning: constant `gpioa_data` should have an upper case name
>   --> src/main.rs:11:7
>    |
> 11 | const gpioa_data: u32 = (0x4001_0800 + 0xc);
>    |       ^^^^^^^^^^ help: convert the identifier to upper case: `GPIOA_DATA`
> 
> error: linking with `rust-lld` failed: exit code: 1
>   |
>   = note: "rust-lld" "-flavor" "gnu" "-L" "/home/oceanfish81/snap/rustup/common/rustup/toolchains/nightly-i686-unknown-linux-gnu/lib/rustlib/riscv32imac-unknown-none-elf/lib" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.19pbfx1e7ppiz22i.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.1gxn96a4ynn6krtc.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.1rrq47tcmo7r1d7q.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.2oywxm1qwyrau2j5.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.36wqpyrxaht0bqyw.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.4g879aqcx3v63xcn.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.hpztk5fc5xtxkyj.rcgu.o" "-o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5" "--gc-sections" "-L" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps" "-L" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/debug/deps" "-L" "/home/oceanfish81/snap/rustup/common/rustup/toolchains/nightly-i686-unknown-linux-gnu/lib/rustlib/riscv32imac-unknown-none-elf/lib" "-Bstatic" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/libr0-c85b5ab54aceb5f9.rlib" "--start-group" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/libpanic_abort-37322ac1e75fee7c.rlib" "/home/oceanfish81/snap/rustup/common/rustup/toolchains/nightly-i686-unknown-linux-gnu/lib/rustlib/riscv32imac-unknown-none-elf/lib/librustc_std_workspace_core-1bd0d3673780d693.rlib" "/home/oceanfish81/snap/rustup/common/rustup/toolchains/nightly-i686-unknown-linux-gnu/lib/rustlib/riscv32imac-unknown-none-elf/lib/libcore-5aec724f1f867fb1.rlib" "--end-group" "/home/oceanfish81/snap/rustup/common/rustup/toolchains/nightly-i686-unknown-linux-gnu/lib/rustlib/riscv32imac-unknown-none-elf/lib/libcompiler_builtins-deaf2d068ddf18c0.rlib" "-Tlink.x" "-Bdynamic"
>   = note: rust-lld: error: no memory region specified for section '.eh_frame'
>           
> 
> error: aborting due to previous error; 7 warnings emitted
> 
> error: could not compile `rust-sipeed-longan-nano`
> 
> Caused by:
[... cut ...] -longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/debug/deps --extern panic_abort=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/libpanic_abort-37322ac1e75fee7c.rlib --extern r0=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/libr0-c85b5ab54aceb5f9.rlib -C link-arg=-Tlink.x` (exit code: 1)
>

advancedwebdeveloper, 2020-11-10:
CC @fanghuaqi @lupyuen @tekkamanninja

Disasm, 2020-11-10:
This problem should already be solved in `riscv-rt` v0.7.2 and v0.8.0. Make sure you use the latest versions of `riscv`, `riscv-rt` and `longan-nano`: they were updated recently.

advancedwebdeveloper, 2020-11-10:
@Disasm , how should I check that?

Disasm, 2020-11-10:
You can run `cargo tree` (you can install it with `cargo install cargo-tree`) and check crate versions displayed

advancedwebdeveloper, 2020-11-10:
@Disasm , I did like this:

> ~/Desktop/rust-sipeed-longan-nano-master$ cargo install cargo-tree
>     Updating crates.io index
>   Downloaded cargo-tree v0.29.0
>   Downloaded 1 crate (15.6 KB) in 0.83s
>   Installing cargo-tree v0.29.0
>   Downloaded structopt-derive v0.4.13
>   Downloaded structopt v0.3.20
>   Downloaded proc-macro-error-attr v1.0.4
>   Downloaded ordermap v0.3.5
>   Downloaded strsim v0.8.0
>   Downloaded fixedbitset v0.1.9
>   Downloaded lazy_static v1.4.0
>   Downloaded serde v1.0.117
>   Downloaded unicode-xid v0.2.1
>   Downloaded version_check v0.9.2
>   Downloaded vec_map v0.8.2
>   Downloaded quote v1.0.7
>   Downloaded unicode-width v0.1.8
>   Downloaded ryu v1.0.5
>   Downloaded itoa v0.4.6
>   Downloaded semver-parser v0.7.0
>   Downloaded atty v0.2.14
>   Downloaded semver v0.9.0
>   Downloaded textwrap v0.11.0
>   Downloaded proc-macro-error v1.0.4
>   Downloaded libc v0.2.80
>   Downloaded bitflags v1.2.1
>   Downloaded syn v1.0.48
>   Downloaded unicode-segmentation v1.6.0
>   Downloaded petgraph v0.4.13
>   Downloaded serde_json v1.0.59
>   Downloaded serde_derive v1.0.117
>   Downloaded proc-macro2 v1.0.24
>   Downloaded heck v0.3.1
>   Downloaded clap v2.33.3
>   Downloaded cargo_metadata v0.9.1
>   Downloaded ansi_term v0.11.0
>   Downloaded anyhow v1.0.34
>   Downloaded 33 crates (1.9 MB) in 1.87s
>    Compiling proc-macro2 v1.0.24
>    Compiling unicode-xid v0.2.1
>    Compiling version_check v0.9.2
>    Compiling syn v1.0.48
>    Compiling serde v1.0.117
>    Compiling libc v0.2.80
>    Compiling ryu v1.0.5
>    Compiling bitflags v1.2.1
>    Compiling serde_json v1.0.59
>    Compiling unicode-width v0.1.8
>    Compiling serde_derive v1.0.117
>    Compiling unicode-segmentation v1.6.0
>    Compiling strsim v0.8.0
>    Compiling ansi_term v0.11.0
>    Compiling vec_map v0.8.2
>    Compiling anyhow v1.0.34
>    Compiling semver-parser v0.7.0
>    Compiling itoa v0.4.6
>    Compiling ordermap v0.3.5
>    Compiling lazy_static v1.4.0
>    Compiling fixedbitset v0.1.9
>    Compiling proc-macro-error-attr v1.0.4
>    Compiling proc-macro-error v1.0.4
>    Compiling textwrap v0.11.0
>    Compiling heck v0.3.1
>    Compiling petgraph v0.4.13
>    Compiling quote v1.0.7
>    Compiling semver v0.9.0
>    Compiling atty v0.2.14
>    Compiling clap v2.33.3
>    Compiling cargo_metadata v0.9.1
>    Compiling structopt-derive v0.4.13
>    Compiling structopt v0.3.20
>    Compiling cargo-tree v0.29.0
>     Finished release [optimized] target(s) in 7m 10s
>   Installing /home/oceanfish81/.cargo/bin/cargo-tree
>    Installed package `cargo-tree v0.29.0` (executable `cargo-tree`)
> 
> 
and
> ~/Desktop/rust-sipeed-longan-nano-master$ cargo tree
> rust-sipeed-longan-nano v0.1.0 (/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master)
> ├── panic-abort v0.3.1
> └── r0 v0.2.2
> 

Here is what I get:

> $ cargo install riscv
>     Updating crates.io index
>   Downloaded riscv v0.6.0
>   Downloaded 1 crate (64.4 KB) in 0.58s
> error: specified package `riscv v0.6.0` has no binaries
> 

> $ cargo install riscv-rt
>     Updating crates.io index
>   Downloaded riscv-rt v0.8.0
>   Downloaded 1 crate (24.4 KB) in 0.72s
> error: specified package `riscv-rt v0.8.0` has no binaries
> 

> $ cargo install longan-nano
>     Updating crates.io index
>   Downloaded longan-nano v0.1.1
>   Downloaded 1 crate (14.1 KB) in 0.45s
> error: specified package `longan-nano v0.1.1` has no binaries
>

advancedwebdeveloper, 2020-11-10:
Looks like I had to add those ones as dependencies, into Cargo.toml.

After updating from

> [package]
> name = "rust-sipeed-longan-nano"
> version = "0.1.0"
> authors = ["David Craven <david@craven.ch>"]
> edition = "2018"
> 
> [dependencies]
> r0 = "0.2.2"
> panic-abort = "0.3.1"

to

> [package]
> name = "rust-sipeed-longan-nano"
> version = "0.1.0"
> authors = ["David Craven <david@craven.ch>"]
> edition = "2018"
> 
> [dependencies]
> r0 = "0.2.2"
> panic-abort = "0.3.1"
> riscv = "0.6.0"
> riscv-rt = "0.8.0"
> longan-nano = "0.1.1"

So it resulted in

> 
> $ cargo build
>     Updating crates.io index
>   Downloaded thread_local v1.0.1
>   Downloaded memchr v2.3.4
>   Downloaded nb v0.1.3
>   Downloaded embedded-hal v0.2.4
>   Downloaded gd32vf103-pac v0.2.0
>   Downloaded vcell v0.1.2
>   Downloaded riscv-target v0.1.2
>   Downloaded bare-metal v0.2.5
>   Downloaded nb v1.0.0
>   Downloaded quote v0.6.13
>   Downloaded regex-syntax v0.6.21
>   Downloaded void v1.0.2
>   Downloaded unicode-xid v0.1.0
>   Downloaded r0 v1.0.0
>   Downloaded syn v0.15.44
>   Downloaded proc-macro2 v0.4.30
>   Downloaded cast v0.2.3
>   Downloaded regex v1.4.2
>   Downloaded aho-corasick v0.7.15
>   Downloaded rand_core v0.4.2
>   Downloaded riscv v0.5.4
>   Downloaded bit_field v0.9.0
>   Downloaded bit_field v0.10.1
>   Downloaded riscv-rt-macros v0.1.6
>   Downloaded rand_core v0.3.1
>   Downloaded gd32vf103xx-hal v0.2.3
>   Downloaded rand v0.5.6
>   Downloaded rustc_version v0.2.3
>   Downloaded 28 crates (1.6 MB) in 1.63s
>    Compiling semver-parser v0.7.0
>    Compiling memchr v2.3.4
>    Compiling lazy_static v1.4.0
>    Compiling regex-syntax v0.6.21
>    Compiling proc-macro2 v0.4.30
>    Compiling unicode-xid v0.1.0
>    Compiling riscv v0.5.4
>    Compiling rand_core v0.4.2
>    Compiling nb v1.0.0
> error[E0463]: can't find crate for `core`
>   |
>   = note: the `riscv32imac-unknown-none-elf` target may not be installed
> 
> error: aborting due to previous error
> 
> For more information about this error, try `rustc --explain E0463`.
> error: could not compile `nb`
> 
> To learn more, run the command again with --verbose.
>

advancedwebdeveloper, 2020-11-10:
>   Running `rustc --crate-name nb --edition=2018 /home/oceanfish81/.cargo/registry/src/github.com-1285ae84e5963aae/nb-1.0.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=ff11b26417734117 -C extra-filename=-ff11b26417734117 --out-dir /home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps --target riscv32imac-unknown-none-elf -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/debug/deps --cap-lints allow -C link-arg=-Tlink.x`
> error[E0463]: can't find crate for `core`
>   |
>   = note: the `riscv32imac-unknown-none-elf` target may not be installed
> 
> error: aborting due to previous error
> 
> For more information about this error, try `rustc --explain E0463`.
> error: could not compile `nb`
> 
> Caused by:
>   process didn't exit successfully: `rustc --crate-name nb --edition=2018 /home/oceanfish81/.cargo/registry/src/github.com-1285ae84e5963aae/nb-1.0.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=ff11b26417734117 -C extra-filename=-ff11b26417734117 --out-dir /home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps --target riscv32imac-unknown-none-elf -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/debug/deps --cap-lints allow -C link-arg=-Tlink.x` (exit code: 1)
> ~/Desktop/rust-sipeed-longan-nano-master$ rustc --crate-name nb --edition=2018 /home/oceanfish81/.cargo/registry/src/github.com-1285ae84e5963aae/nb-1.0.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=ff11b26417734117 -C extra-filename=-ff11b26417734117 --out-dir /home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps --target riscv32imac-unknown-none-elf -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps -L dependency=/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/debug/deps --cap-lints allow -C link-arg=-Tlink.x
> {"message":"can't find crate for `core`","code":{"code":"E0463","explanation":"A plugin/crate was declared but cannot be found.\n\nErroneous code example:\n\n```compile_fail,E0463\n#![feature(plugin)]\n#![plugin(cookie_monster)] // error: can't find crate for `cookie_monster`\nextern crate cake_is_a_lie; // error: can't find crate for `cake_is_a_lie`\n```\n\nYou need to link your code to the relevant crate in order to be able to use it\n(through Cargo or the `-L` option of rustc example). Plugins are crates as\nwell, and you link to them the same way.\n"},"level":"error","spans":[{"file_name":"/home/oceanfish81/.cargo/registry/src/github.com-1285ae84e5963aae/nb-1.0.0/src/lib.rs","byte_start":0,"byte_end":0,"line_start":1,"line_end":1,"column_start":1,"column_end":1,"is_primary":true,"text":[],"label":"can't find crate","suggested_replacement":null,"suggestion_applicability":null,"expansion":null}],"children":[{"message":"the `riscv32imac-unknown-none-elf` target may not be installed","code":null,"level":"note","spans":[],"children":[],"rendered":null}],"rendered":"\u001b[0m\u001b[1m\u001b[38;5;9merror[E0463]\u001b[0m\u001b[0m\u001b[1m: can't find crate for `core`\u001b[0m\n\u001b[0m  \u001b[0m\u001b[0m\u001b[1m\u001b[38;5;12m|\u001b[0m\n\u001b[0m  \u001b[0m\u001b[0m\u001b[1m\u001b[38;5;12m= \u001b[0m\u001b[0m\u001b[1mnote\u001b[0m\u001b[0m: the `riscv32imac-unknown-none-elf` target may not be installed\u001b[0m\n\n"}
> {"message":"aborting due to previous error","code":null,"level":"error","spans":[],"children":[],"rendered":"\u001b[0m\u001b[1m\u001b[38;5;9merror\u001b[0m\u001b[0m\u001b[1m: aborting due to previous error\u001b[0m\n\n"}
> {"message":"For more information about this error, try `rustc --explain E0463`.","code":null,"level":"failure-note","spans":[],"children":[],"rendered":"\u001b[0m\u001b[1mFor more information about this error, try `rustc --explain E0463`.\u001b[0m\n"}
>

Disasm, 2020-11-14:
> error[E0463]: can't find crate for core
|
= note: the riscv32imac-unknown-none-elf target may not be installed
This means that you need to install the `riscv32imac-unknown-none-elf` target. You can do this by running
`rustup target add riscv32imac-unknown-none-elf`.

Disasm, 2021-07-26:
Closing as the problem seems to be resolved.

closed 2021-07-26 by Disasm