rust-embedded/riscv #60: RVXXI registers

Skallwar, 2020-11-23:
Is there a reason why RVXXI registers are not available? If this is just an issue of implementing it, I will gladly do it

[image]

luojia65, 2021-02-16:
In my opinion these registers are used by programming code, and operations on common registers should be controlled by compiler or inline assembly only. It's not a common case that we should operate these registers manually.

(In some cases `gp` and `tp` are used when developing OS kernel, but inline assembly should handle with them)

romancardenas, 2023-09-21:
I agree with @luojia65, these registers are compiler/inline assembly territory. Closing the issue.

closed 2023-09-21 by romancardenas