rust-embedded/riscv #74: Implementing PMP registers as an array?

jwnhy, 2021-07-01:
Is it a good idea to implement a group of registers as an array or other structure that allows index access?

Useful Scenario:
```rust
 impl Region {
     fn to_napot(&self) -> usize {
         if self.size <= 2 {
            self.addr
         } else {
             /*1...11111011111...1*/
             let mask = !(1 << (self.size - 2)); ‣mask: usize
             let mask = (mask << (XLEN-self.size+1)) >> (XLEN-self.size+1);
             self.addr | mask
         }
     }
     fn to_tor(&self) -> (usize, usize) {
         (self.addr, self.addr + self.size)
     }
     // Here we want to access pmp registers by index.
     fn enforce(&self, number: usize) {
         // pmpcfg[number].write(config)
     }
 }

 pub struct MemoryLayout {
     /* at most 16 pmp region is allowed */
     regions: [Region; 16]
 }
```
Or is there a way to access these register through a `i:usize -> pmpcfgi` process?
I'm not sure if macro can do this.

dkhayes117, 2021-07-17:
I address this in a different way in PR #58.  Feel free to look it over and see if you have any suggestions.

jwnhy, 2022-03-04:
As it has been addressed, closing

closed 2022-03-04 by jwnhy