rust-embedded/riscv #85: "cannot link object files with different floating-point ABI" rustc 1.56.0-nightly (2021-08-06)

o8vm, 2021-08-07:
It seems that the latest nightly build has the same issue as #69.

output:

```
  = note: rust-lld: error: /.../riscv64gc-unknown-none-elf/debug/deps/libriscv-c8056877110d3c7c.rlib(riscv.o): cannot link object files with different floating-point ABI

error: could not compile `project` due to previous error
```

rustup show:

```
active toolchain
----------------

nightly-x86_64-unknown-linux-gnu (overridden by '/home/o8/code/poctx/rust-toolchain')
rustc 1.56.0-nightly (5ad7389bd 2021-08-06)
```

Disasm, 2021-08-07:
Are you using the latest `riscv` crate release?

o8vm, 2021-08-07:
oh sorry. I updated riscv crate to the latest 0.7.0 and that fixed the error. 
Thank you very much.

closed 2021-08-07 by o8vm