rust-embedded/riscv #92: RISCV sdk with RUST support

AchyuthAGNA5675, 2022-01-18:
Hello everyone,am trying to build SDK for RISC-V which supports RUST lang,so i can cross compile a file from my x86-linux-machine to RISC-V board(). Am a Newbie,can i get some help please??? i tried to add rust binary and trying to build SDK with rust support. but i couldn't accomplish it.

My goal is to cross-compile a rust file from x86-linux-Machine to RISC-V board. i already have SDK built for RISC-V sources,is it possible to add RUST support to it?

almindor, 2022-01-18:
Which board are we talking? The target depends on RISC-V feature sets for the board. You can see a good setup example at https://github.com/riscv-rust/riscv-rust-quickstart

orangecms, 2022-11-09:
There is no need for an SDK. Just use Rust's tooling.

For any target, you can use `rustup` to install it.
Check out https://rustup.rs/ for instructions.

Here is an example app set up for targeting RISC-V:
https://github.com/orangecms/hello-linux-rv64-rs

Essentially, the `cargo build` command gets the `--target` flag; e.g., for Linux with GNU:
```
cargo build --release --target riscv64gc-unknown-linux-gnu
```

almindor, 2022-11-09:
Please re-open if there is still something unclear.

closed 2022-11-09 by almindor