rust-embedded/riscv #97: Linking fails when using this crate as a dependency

JarredAllen, 2022-02-26:
I'm attempting to compile a rust crate using this library (targeting a bare-metal build on rv32imac), and I get this error during the linking step:
```
ld: error: target/riscv32imac-unknown-none-elf/debug/libcrate.a(riscv.o): conflicting priv spec version (major/minor/revision).
ld: error: target/riscv32imac-unknown-none-elf/debug/libcrate.a(riscv.o): conflicting priv spec version (major/minor/revision).
ld: failed to merge target specific data of file target/riscv32imac-unknown-none-elf/debug/libcrate.a(riscv.o)
```
This error is caused by this crate specifically, which I know because it only started when I imported this crate, and I can build fine with other crates linked in, it is specifically this crate as a dependency that causes this error to appear. Is this behavior something known about? I can't seem to find anything helpful in online search results for this error message.

JarredAllen, 2022-02-26:
I think so. The project was set up with the default [cargo-pio](https://github.com/ivmarkov/embuild/tree/HEAD/cargo-pio) for the SparkFun RED-V Thing Plus, because I'm trying to build targeting that chip, so it includes the default includes for such a project, which I think includes a few object files compiled from C (though I don't use them for anything and, looking at completed builds without this crate imported, they're entirely garbage-collected out).

I just realized that having an example of the build failing might be helpful, so I made a minimal example of things going wrong here: https://github.com/JarredAllen/RISC-V-and-Cargo-Pio-Example

Also, I just noticed while making it that if this crate is included as a dependency but nowhere used, then the linker will garbage-collect everything from this crate out of the final binary and will not report an error. It is only when I use something from this crate that the error happens. I don't know enough about linkers to know if that means anything about the source of the error, but I thought it might be worth mentioning.

Disasm, 2022-02-27:
I tried to build this example with stable 1.59.0 and nightly compilers and observed no problems. Could you describe your environment? Rust compiler version, RISC-V gcc compiler version, OS version, build commands.

JarredAllen, 2022-02-27:
I've observed this on rust nightly (both `rustc 1.61.0-nightly (532d3cda9 2022-02-23)` and `rustc 1.61.0-nightly (10cc7a6d0 2022-02-26)` (the latest nightly release), as well as some older versions that I no longer have record of). I'm on Ubuntu 20.04.

I installed the RISC-V gcc toolchain through cargo-pio and just used it's default installation, using the commands I put in the repo's readme. The gcc compiler version is `riscv64-unknown-elf-gcc (SiFive GCC 8.3.0-2019.08.0) 8.3.0` and the ld version is `GNU ld (SiFive Binutils 2.32.0-2019.08.0) 2.32`. To build, I just run `cargo pio build` (also matching the command in the readme).

romancardenas, 2024-05-15:
I think this issue no longer applies, as we now use inline assembly. Feel free to re-open it in case I'm wrong.

closed 2024-05-15 by romancardenas