rust-embedded/riscv #100: Support for riscv32imc.

mgaggero, 2022-04-19:
Hi,
    compiling for Xtensa ESP32-C3 result in an error for a missing file:

```
Caused by:
  process didn't exit successfully: `target/debug/build/riscv-340f5de0c5d11624/build-script-build` (exit status: 101)
   --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', /riscv-0.7.0/build.rs:23:10
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

If I modify the code to print out the name of the missing file it says it is the `bin/riscv32imc-esp-espidf.a`.
In the `bin/` folder a binary for riscv32imc is missing.
If I copy the _riscv32ic.a_ to _riscv32imc.a_ the program compiles successfully and it seems to work.

Massimo.

almindor, 2022-04-20:
Hmm this is a bit odd. My understanding is that `build.rs` should use `retain_extension` and reduce `riscv32imc` into `riscv32ic`.

What is your `TARGET` env var exactly?

almindor, 2022-04-20:
> I put a **println!** line on build.rs:
> 
> ```
> fn main() {
>     let target = env::var("TARGET").unwrap();
>     println!("================================================================================");
>     println!("TARGET = {}", target);
>     println!("================================================================================");
> ```
> 
> and it says:
> 
> `TARGET = riscv32imc-esp-espidf `

Using that, if I call
```rust
use riscv_target::Target;

fn main() {
    let raw = "riscv32imc-esp-espidf";
    let mut target = Target::from_target_str(raw);
    target.retain_extensions("ifdc");

    let target = target.to_string();
    let target = target.split("-").next().unwrap();

    println!("{}", target);
}
```

I get `riscv32ic` which should be ok, so I'm a bit confused where the issue is. The code I'm referencing here is from [this build.rs file](https://github.com/rust-embedded/riscv/blob/master/build.rs#L13-L19).

Disasm, 2022-04-20:
I think the relevant fix for ESP32 (https://github.com/rust-embedded/riscv/pull/93) is not released yet.

almindor, 2022-04-20:
Ah good point, I was testing on master. @mgaggero could you try with latest master build of `riscv` crate to see if it all works out?

mgaggero, 2022-04-20:
Ok, using the master branch compiles successfully. The problem seems to relies on the version 0.7.0 required by the crate _critical-section_. I've noticed that a new version 0.8.0 has been released. I'll try it and possibly ask to update the dependencies' versions of this last. Thank you!

closed 2022-04-20 by mgaggero