rust-embedded/riscv #116: The `critical_section` implementation is wrong

Tnze, 2022-10-16:
These two lines is not atomic:

https://github.com/rust-embedded/riscv/blob/d9c60763b8fb53097ad1d95ff99c1eabe216dc34/src/critical_section.rs#L11-L13

and sould be replaced by a single `csr` instruction:
```rust
let mut mstatus: usize;
asm!("csrrci {}, 0x300, 0b100", out(reg) mstatus);
core::mem::transmute::<_, Mstatus>(mstatus).mie()
```
or it could be interrupted between reading status and disabling the interrupt.

Tnze, 2022-10-16:
To prevent from using `core::mem::transmute`, we should make the `Mstatus.bits` public:
```rust
let mut mstatus: Mstatus;
asm!("csrrci {}, 0x300, 0b100", out(reg) mstatus.bits);
mstatus.mie()
```

almindor, 2022-10-16:
Good find. Thanks, pushed PR, please review. I've decided to keep the bits `pub(crate)` only to not break the API

almindor, 2022-10-16:
Hmm actually we can't make the `Mstatus::bits` `pub(crate)` since it'd allow instantiation which leads to UB.

commit ed254c7d (2022-10-16): fix atomicity of critical section, fixes #116
commit f847eba2 (2022-10-16): fix atomicity of critical section, fixes #116
commit 462b5509 (2022-10-19): Merge #117

117: fix atomicity of critical section, fixes #116 r=dkhayes117 a=almindor

This changes `SingleHartCriticalSection::acquire` to be done in an atomic way.

Fixes #116

NOTE: no changelog since it fixes a small bug on unreleased change
PR #117 (2022-10-19): fix atomicity of critical section, fixes #116
This changes `SingleHartCriticalSection::acquire` to be done in an atomic way.

Fixes #116

NOTE: no changelog since it fixes a small bug on unreleased change
closed 2022-10-18 by Tnze