rust-embedded/riscv #130: `sip` register set / clear functions

PitiBouchon, 2023-04-20:
There are many functions for the `sie` register to set and clear bits but nothing for the `sip` register which can only be read
I don't know why but it would be super useful (for example if you want to clear the STIP bit to acknowledge a supervisor timer interrupt

romancardenas, 2023-04-21:
I'll check this in depth, but I'm 99% sure that usually you don't clear the interrupt pending flags directly from the `sip` register. Instead, you interface the interrupt controller of your particular target (e.g., CLINT, PLIC, ...).

romancardenas, 2023-04-21:
Hmm, it seems weird to me that [`mip`](https://github.com/rust-embedded/riscv/blob/ec99d035d62abaf2e6053c263e8a58a1186d1bbd/src/register/mip.rs#L74) allows you to clear/set the interrupt pending bits whereas [`sip`](https://github.com/rust-embedded/riscv/blob/ec99d035d62abaf2e6053c263e8a58a1186d1bbd/src/register/sip.rs#L55) does not... @rust-embedded/riscv any insights about this difference?

PitiBouchon, 2023-04-21:
> I'll check this in depth, but I'm 99% sure that usually you don't clear the interrupt pending flags directly from the `sip` register. Instead, you interface the interrupt controller of your particular target (e.g., CLINT, PLIC, ...).

Okay thx, I misunderstood something

romancardenas, 2023-04-21:
No problem! Check [this old issue](https://github.com/rust-embedded/riscv/issues/62), maybe it helps you with your use case :)

closed 2023-04-21 by PitiBouchon