rust-embedded/riscv #146: Platform specific extensions

SoulSharer, 2023-10-09:
[image]

Are there any plans to provide support for platform specific extensions in `riscv` crate?
Reading through *RISC-V Privileged Architectures* specification I see that there are cases where, for example, a `3.1.15 Machine Cause Register` has fields for platform and custom use.

Was wondering if it would be possible to extend enums `Interrupt` and `Exception` from outside of the `riscv` crate itself.
The best I could came up thus far is adding "PlatformSpecific1-32" fields to `Interrupt` with user-defined conversion (`TryFrom<Interrupt> for PlaftormInterrupt`) where applicable. (same goes for `Exception`.

romancardenas, 2023-10-14:
Hi! I don't think we have plans to add this functionality right now. If you think this feature is useful, feel free to open a PR/RFC and we can discuss it further :)

In my opinion, we should look at how `cortex-m` deals with platform-specific interrupts and follow their approach.

romancardenas, 2024-05-04:
While working on #200, I've been thinking on how to provide this functionality. I propose the following approach:

- Remove weak symbols in linker file (i.e., the `PROVIDE` directives) for interrupt and exception sources
- Provide procedural macros for defining the interrupt and exception sources of a platform. These macros would be in charge of generating all the symbols related to interrupts and exceptions (e.g., `_start_trap_rust`, `__INTERRUPTS`, `__EXCEPTIONS`, `_vector_table`...). Additionally, the code will provide weak symbols to all these interrupts using global inline assembly.
- By default, `riscv-rt` will use this new macro to generate a code that would be equivalent to its current version. I expect this change not to be a breaking change.
- A new feature would be able to opt-out the standard implementation. In this way, a new `my-riscv-specific-rt` runtime crate would have `riscv-rt` as a dependency, re-exporting everything and calling to the interrupt macro accordingly to the exception codes of the target.

closed 2024-10-17 by romancardenas