rust-embedded/riscv #148: FCSR operations generally cannot be used from Rust

RalfJung, 2023-11-04:
This crate seems to [provide operations](https://github.com/rust-embedded/riscv/blob/master/src/register/fcsr.rs) that can change the floating-point rounding mode and read the accrued exceptions flags. These operations generally cannot be used soundly from Rust; see https://github.com/rust-lang/stdarch/pull/1478 for more explanation.

It would be good to clarify in the documentation when and how exactly these functions are meant to be used. As things stand, calling `set_rounding_mode()` will be immediate UB in almost all contexts, and `read().fflags()` will return an unreliable value.

romancardenas, 2023-11-06:
Thanks for this issue! I see that you are removing all the floating point configuration-related functions from the standard library. Thus here I see two options: either we leave all the functions `unsafe` and properly documenting them, or we just remove them following the standard implementation.

RalfJung, 2023-11-06:
I originally planned to deprecate them, which was what we did on x86. But it was pointed out that they are still unstable so we might as well remove them.

Exposing some unsafely is an option, but the question here is what the expected use-case looks like. The one example I can imagine is implementing context switching, where you have to save/restore the FCSR of the userland process. But that's likely better done in inline assembly, there's going to be a bunch of inline assembly in that code anyway.

For regular userland code, there isn't really a way to use these functions correctly. However, if they are removed, documentation should probably be added that tells users why they aren't there, and how to use inline assembly instead.

closed 2023-11-08 by MabezDev