rust-embedded/riscv #153: Linker relocation issue (QEMU / OpenSBI)

chutchinson, 2023-11-28:
I am experimenting with the RISC-V ISA via QEMU emulation, and I am having trouble figuring out how to link in the _heap_size symbol appropriately, so that I can follow through with heap allocation. I followed the **riscv-rt** documentation closely. I am attempting to use QEMU with OpenSBI, but I am not experienced with linker configuration. Here is my memory layout specification:

```text
MEMORY
{
    RAM : ORIGIN = 0x80200000, LENGTH = 0x8000000
    FLASH : ORIGIN = 0x20000000, LENGTH = 16M
}

REGION_ALIAS("REGION_TEXT", RAM);
REGION_ALIAS("REGION_RODATA", RAM);
REGION_ALIAS("REGION_DATA", RAM);
REGION_ALIAS("REGION_BSS", RAM);
REGION_ALIAS("REGION_HEAP", RAM);
REGION_ALIAS("REGION_STACK", RAM);
```

I understand that to mean that there is 16M of reserved flash memory at the beginning of the address space (OpenSBI), followed by RAM address space. I would expect all of the memory regions to fit within RAM since the FLASH space is reserved by QEMU OpenSBI firmware.

I have the following example program:

```rust
#![no_std]
#![no_main]

extern crate panic_halt;

mod uart;
mod print;

use riscv_rt::entry;

use uart::Uart;

extern "C" {
    static _heap_size: u8;
}

#[entry]
fn __start() -> ! {
    let heap_size = unsafe { &_heap_size as *const u8 as usize };
    println!("hello world");
    println!("heap size = {}", heap_size);
    loop {}
}
```

The problem is that the linker cannot relocate **_heap_size** provided by riscv-rt and I do not understand why. I tried changing the code model to **medium**:

```text
relocation R_RISCV_PCREL_HI20 out of range: -524800 is not in [-524288, 524287]; references _heap_size
```

vilukissa68, 2024-06-19:
We ran into this or a similar issue with @hegza and were able to solve it by setting `riscv64-unknown-elf-ld` as linker in `config.toml`
```
linker = "riscv64-unknown-elf-ld"
```

alistair23, 2024-06-19:
Why use OpenSBI though? Are you trying to write S-mode software?

hegza, 2024-06-19:
In our case it wasn't OpenSBI but an application thay needed access to a memory located in the high bits.