rust-embedded/riscv #154: Using this as a library?

hyperswine, 2023-11-28:
Is it possible to use this as a "just works" library without extra copying/config or is it meant to be used as a template?

romancardenas, 2024-02-15:
I'm closing this issue as stale, as I believe the problem was that the linker arguments were not provided. Feel free to re-open it if you consider it necessary.

closed 2024-02-15 by romancardenas

xobs, 2025-11-20:
I believe this issue should be reopened.

There is a directory called `examples`, and if you use the examples they don't work and produce the error mentioned.

The examples should be fixed so that they work, and documentation should be added on how to use it.

Neither [the project's README.md](https://github.com/rust-embedded/riscv/blob/master/riscv-rt/README.md) nor the [docs.rs documentation](https://docs.rs/crate/riscv-rt/latest) give any indication on what is needed.

romancardenas, 2025-11-20:
I'm currently under very tight deadlines and won't be able to work on this. I will try to get back ASAP.