rust-embedded/riscv #157: Implement FPU initialization

Disasm, 2023-11-28:


mini-ninja-64, 2023-12-06:
I am happy to have a go at implementing this, what would be the preferred approach for setting FPU as enabled?

a piece of config similar to the m-mode / s-mode switch that the crate currently uses, or something more like if in m-mode use misa config to auto detect FPU presence?

I feel like I read somewhere before misa should be avoided in favour of device tree so maybe it would not make sense to use it for this 🤔 especially as in s-mode we would have to call opensbi or something wierd like that for reading misa and then sstatus for actually setting fs.

romancardenas, 2023-12-06:
If your target supports the `f` extension, you could use a configuration gate like `#[cfg(riscvf)]` (check [this](https://github.com/rust-embedded/riscv/blob/c579937a17770e26754e448f52e291d511ec99ce/riscv-rt/build.rs#L75)).

We do something similar [here](https://github.com/rust-embedded/riscv/blob/c579937a17770e26754e448f52e291d511ec99ce/riscv-rt/src/asm.rs#L100) for targets supporting integer multiplication.

PR #166 (2023-12-19): `riscv-rt`: Implement FPU initialization
This PR adds FPU initialization when compiling with the `f` or `d` extensions, for the implementation of this I have updated the `parse_target` function from `build.rs` to support additional extensions, this is required until this crate is [updated to rust 1.75](https://github.com/rust-lang/rust/pull/116485) at which point we can use `target_feature`.

I added a little macro to help with zeroing registers, if it is preferable though happy to inline the zeroing of registers

I am new to rust & risc-v in general so apologies if there is any obvious mistak [... cut]
closed 2023-12-19 by romancardenas