rust-embedded/riscv #173: Release time

romancardenas, 2024-01-10:
It's been months of lots of changes, and I think it is time to pack a new release for all the crates in this workspace.

I would love to have #164 merged so we can make a release for all the crates at once. What do you think, @rust-embedded/riscv ?

If we are all OK with this, I'll set up a PR with proper version numbers and, once it is merge to `master`, I can publish the releases

georglauterbach, 2024-01-13:
Indeed; I am currently stuck on [`28b916d`@rust-embedded/riscv](https://github.com/rust-embedded/riscv-rt/tree/28b916dc400caef2b3bfd4c5e66130a162e21f26) (due to me requiring the most recent linker changes) and with the move to this repository, I can't seem to include the `riscv-rt` crate in `Cargo.toml` via `git = "...", rev = "..."`. Having a release is the only way of updating for me.

romancardenas, 2024-01-14:
@georglauterbach thanks for your comment! Let's see if we can do a quick release today or within a few days

closed 2024-01-15 by romancardenas