rust-embedded/riscv #175: `riscv-rt`: building for riscv32imac-unknown-none-elf gives an error with the mul instruction

daniestevez, 2024-01-22:
I found this problem first in https://github.com/daniestevez/galileo-osnma/issues/23, but the following self-contained example replicates the problem: https://github.com/daniestevez/riscv/tree/debug-mul-build/riscv-rt/test-build

When this example is built with `cargo build --release`, I get the following:

```
   Compiling proc-macro2 v1.0.78
   Compiling unicode-ident v1.0.12
   Compiling syn v1.0.109
   Compiling riscv-rt v0.12.0 (/home/daniel/riscv/riscv-rt)
   Compiling riscv v0.11.0 (/home/daniel/riscv/riscv)
   Compiling embedded-hal v1.0.0
   Compiling critical-section v1.1.2
   Compiling test-build v0.1.0 (/home/daniel/riscv/riscv-rt/test-build)
   Compiling panic-halt v0.2.0
   Compiling quote v1.0.35
   Compiling riscv-rt-macros v0.2.1 (/home/daniel/riscv/riscv-rt/macros)
error: instruction requires the following: 'M' (Integer Multiplication and Division) or 'Zmmul' (Integer Multiplication)
mul t0, t2, t0
^
error: instruction requires the following: 'M' (Integer Multiplication and Division) or 'Zmmul' (Integer Multiplication)
mul t0, t2, t0
^
    Finished release [optimized] target(s) in 2.11s

```

Despite the error, an executable is produced.

If I build without `--release` instead, I get:

```
   Compiling proc-macro2 v1.0.78
   Compiling unicode-ident v1.0.12
   Compiling syn v1.0.109
   Compiling riscv-rt v0.12.0 (/home/daniel/riscv/riscv-rt)
   Compiling riscv v0.11.0 (/home/daniel/riscv/riscv)
   Compiling critical-section v1.1.2
   Compiling embedded-hal v1.0.0
   Compiling test-build v0.1.0 (/home/daniel/riscv/riscv-rt/test-build)
   Compiling panic-halt v0.2.0
   Compiling quote v1.0.35
   Compiling riscv-rt-macros v0.2.1 (/home/daniel/riscv/riscv-rt/macros)
    Finished dev [unoptimized + debuginfo] target(s) in 2.51s
```

romancardenas, 2024-01-22:
I noticed this error. It looks like a weird issue between the Rust compiler and LLVM. Perhaps it does not propagate the extension flags correctly?

In any case, it produces a working binary, so it is not critical (it is annoying, though)

daniestevez, 2024-01-22:
I forgot to mention that this didn't happen with an earlier version of `riscv-rt` (or maybe it was with an earlier version of `rustc`; I forgot of what I updated when this started happening). I agree that the problem is not at all critical.

romancardenas, 2024-01-23:
The previous version of `riscv-rt` used pre-compiled blobs linked at compile time. Now, we use inline assembly, and the Rust compiler includes the assembly code in the build process. I'll ask the rest of the Rust Embedded working group about this issue, they can suggest where to open an issue.

romancardenas, 2024-01-23:
#176 provides a provisional patch to remove these spurious errors. It will do the trick while LLVM solves this issue