rust-embedded/riscv #189: Assembly algorithm for RAM init incompatible with upcoming RVE extension (future proofing)

hegza, 2024-02-23:
The RAM init function at https://github.com/rust-embedded/riscv/blob/f5a2da9a05ebe710d8fc24b38939305de37ecf99/riscv-rt/src/asm.rs#L152-L158 is incompatible with (as of yet, unratified) RV32E base ISA.

The algorithm uses temporaries t0--t3 to copy data from from flash to RAM. However, the RV32E removes registers `x16`--`x31`, which includes `t3` / `x28`, therefore making this algorithm unavailable on RVE.

## Proposals for a solution

1. We could pick another register in range `x0--x16` in place of `t3` to make the code compatible between RVI and RVE
2. We could `#[cfg(riscve)]` the algorithm to use another register on RV32E.

romancardenas, 2024-02-23:
I personally prefer using another register that fits all the specifications

romancardenas, 2024-12-12:
Solved in #243

closed 2024-12-12 by romancardenas