rust-embedded/riscv #209: All the CSR write operations should be unsafe by default

jsgf, 2024-05-14:
In general we should assume that writing to CSRs could do something that potentially violates the Rust abstract model.

Macros like `write_csr_as` and `write_csr_as_usize` should at least default to `unsafe`, and maybe have an option to make a safe variant on a CSR by CSR basis.

romancardenas, 2024-05-14:
As I always do, I checked the [`cortex-m`]/(https://github.com/rust-embedded/cortex-m/blob/4395baee03d57a4e1763c427073da5f33ec3b17e/cortex-m/src/register/lr.rs#L18) crate and... they use the same fashion you propose 😁 

In other words, I am in favor of your proposal.

alistair23, 2024-06-05:
I agree. Writing to RISC-V CSRs can cause all sorts of `unsafe` behaviour. There are a few that probably aren't `unsafe` but they need some reasoning about why they aren't. `unsafe` by default is the way to go

PR #251 (2025-02-08): Make all CSR writes unsafe by default
Closes #209 

**Work in Progress!** I added a `safe` pattern to opt out of unsafety in cases where we consider there to be no safety issues. I will list all the different registers where we use this macro and nominate some to maintain write safety.
closed 2025-02-08 by romancardenas