rust-embedded/riscv #248: `riscv-rt`: ABI incompatibility

rslawson, 2024-12-11:
'm working on a project right now that's using a RISCV-32 core with the I, M, F, and C extensions. I know it wouldn't be too difficult to define a custom target for this, but project CI stuff would make actually doing that very difficult (I don't understand Nix). As such, I'm using `riscv32imc` with `--target-features=+f`. However, this causes something of an issue - this target uses the `ilp32` ABI, but `riscv-rt` sees the `riscvf` cfg flag and thus puts some FP instructions into global assembly, which requires `ilp32f` (as far as I can tell?). Thus I get 33 errors of things like
```
error: instruction requires the following: 'F' (Single-Precision Floating-Point)
   |
note: instantiated into assembly here
  --> <inline asm>:99:1
   |
99 | fscsr x0
   | ^

error: instruction requires the following: 'F' (Single-Precision Floating-Point)
    |
note: instantiated into assembly here
   --> <inline asm>:100:5
    |
100 |     fmv.w.x f0, x0
    |     ^
```
So then the question is: should I just bite the bullet and figure out Nix well enough to get it happy enough to accept a custom `riscv32imfc-uknown-none-elf.json`, or would it be acceptable for me to add a `no-float-init = []` feature to the crate that would mask out those `global_asm!` invocations?

rslawson, 2024-12-11:
Ah, sorry, looks like this can actually be fixed by adjusting the provisional patch at `asm.rs` line 22. I'll PR this soon.

romancardenas, 2025-06-19:
Hi @rslawson ! Any news on this?

rslawson, 2025-06-19:
Not since last working on this sometime last year, no. At least I haven't had any issues with it since then, so I think I can close it for now. I can always reopen this later if the same issue comes up again later.

closed 2025-06-19 by rslawson