rust-embedded/riscv #260: riscv rt linker issue

yashwanthsinghm, 2025-01-27:
We have used riscv_rt in our current projects and encountered a couple of issues:

Cargo release builds are happening but when we compile for debug its giving following error.
-     Linker Issue:  ``` rust-lld: error: <internal»:(.eh_frame+0x40): relocation R_RISCV_32_PCREL out of range: 2147486966 is not in I-2147483648, 2147483 6471; references ```

 github : [Strange21/RISCV: software for RISCV](https://github.com/Strange21/RISCV)

In-order to reproduce the error 

1. cd riscv-hal
2. cargo build --example gptimer

Issue 
```
error: linking with 'rust-lld' failed: exit status: 1
= note: "rust-lld" "-flavor" "gnu" "/var/folders/bn/73c2thl17734d902lypvxbpc0000gn/T/rustcf4Z0vW/symbols.0" "<2 object files omitted»" "-
-as-needed" "-Bstatic" "/Users/IKL2KOR/Desktop/shakti_riscv_hal/examples/gpio_applns/target/riscv64imac-unknown-none-elf/debug/deps/{libshak ti_riscy_hal-e9fb2d117581f3a1. rlib, libtock_registers-dd0b6693e8c40f2b.rlib, libriscv_rt-a141fe7383d82ded. rib, libriscv-bc6c60484b95b10b.rlib,
libcritical_section-e493a1cc7896740d. rlib, libriscv_pac-a4f795429843822a. rlib, libembedded_hal-ab4e22761e14a865.rlib}.rlib" "<sysroot>/lib/rus
tlib/riscv64imac-unknown-none-elf/lib/{librustc_std_workspace_core-*,libcore-*,libcompiler_builtins-*}.rlib" "-Bdynamic" "-z".
"noexecstack"
"-L" "/Users/IKL2KOR/Desktop/shakti_riscv_hal/examples/gpio_applns/target/riscv64imac-unknown-none-elf/debug/build/riscv-rt-5a95d4c5a1c18589 /out" "-L" "/Users/IKL2KOR/Desktop/shakti_riscv_hal/examples/gpio_applns/target/riscv64imac-unknown-none-elf/debug/build/shakti_riscv_hal-8e
= note: rust-lld: error: «internal»: (-eh_frame+0x1c): relocation R_RISCV_32_PCREL out of range: 2147486584 is not in I-2147483648, 2147483
6471; references '.LO '
»> defined in /Users/IKL2KOR/. rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcore
-9699b3174c9bc206. rlib(core-9699b3174c9bc206, core. c4f5009eca0c9ae-cgu.0.rcgu.o)
rust-lld: error: <internal»:(.eh_frame+0x40): relocation R_RISCV_32_PCREL out of range: 2147486966 is not in I-2147483648, 2147483
6471; references
»> defined in /Users/IKL2KOR/. rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcore
-9699b3174c9bc206. rlib(core-9699b3174c9bc206. core. c4f5009eca@c9ae-cgu.0.rcgu.o)
rust-lld: error: <internal>: (.eh_frame+0x64): relocation R_RISCV_32_PCREL out of range: 2147487738 is not in I-2147483648, 2147483
647]; references 'LO '
»> defined in /Users/IKL2KOR/. rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcore
-9699b3174c9bc206. rlib(core-9699b3174c9bc206.core.c4f5009ecac9ae-cgu.0.rcgu.o)
rust-lld: error: «internal>: (.eh_frame+0x84): relocation R_RISCV_32_PCREL out of range: 2147487736 is not in I-2147483648, 2147483
6471; references
'.LO'
»> defined in /Users/IKLKOR/. rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcore
-9699b3174c9bc206. rlib(core-9699b3174c9bc206. core. c4f5009eca@c9ae-cgu.0.rcgu.o)
rust-lld: error: ‹internal>: (.eh_frame+0xa4): relocation R_RISCV_32_PCREL out of range: 2147487738 is not in I-2147483648, 2147483
6471; references
'LO'
›>› defined in /Users/IKL2KOR/. rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcore
-9699b3174c9bc206. rlib(core-9699b3174c9bc206. core. c4f5009eca0c9aбe-cgu.0.rcgu.o)
rust-lld: error: «internal»:(.eh_frame+0xc4): relocation R_RISCV_32_PCREL out of range: 2147487782 is not in I-2147483648, 2147483
647]; references '.LO '
»› defined in /Users/IKLKOR/. rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcore
-9699b3174c9bc206. rlib(core-9699b3174c9bc206. core. c4f5009eca@c9a6e-cgu.0.rcgu.o)
rust-lld: error: internal»: (.eh_frame+0xe4): relocation R_RISCV_32_PCREL out of range: 2147487808 is not in I-2147483648, 2147483
647]; references 'LO '
›>› defined in /Users/IKL2KOR/. rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/riscv64imac-unknown-none-elf/lib/libcore
-9699b3174c9bc206. rlib(core-9699b3174c9bc206. core. c4f5009eca0c9a6e-cgu.0.rcgu.o)

```

romancardenas, 2025-01-27:
Related: #153

romancardenas, 2025-01-27:
I was able to reproduce your issue on my machine. I then changed `riscv-rt` 13.0 to the current version in GitHub and it compiled without issues. Not sure what the problem is, we need to study it more deeply.

yashwanthsinghm, 2025-01-28:
Hi @romancardenas 

If you look into cargo.toml we have already updated  riscv_rt to 13.0, **cargo build -r --example gptimer** works for us which is a release build. But **cargo build --example gptimer** is giving above linker issue.

It is not just in this project but this is reflected in other projects as well, for example you can check out this as well.

```GitHub : https://github.com/yashwanthsinghm/shakti_riscv_hal```

```cargo build --example gpio_applns ```

yashwanthsinghm, 2025-01-28:
Thank you @romancardenas It works. We have tested it with two different projects using the GitHub master branch. You can proceed to publish it

yashwanthsinghm, 2025-01-29:
Hi @romancardenas ,

new issue encountered when i try to use semihosting with riscv_rt, please check out this. 

git : https://github.com/imrank03/riscv_semi.git

its just a basic hprintln!, giving a linker issue is it something to do with riscvrt? 

```    Compiling riscv_semi v0.1.0 (/home/yash/Desktop/riscv_semi)
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/yash/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/yash/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" VSLANG="1033" "rust-lld" "-flavor" "gnu" "/tmp/rustcrqWxx0/symbols.o" "<1 object files omitted>" "--as-needed" "-Bstatic" "/home/yash/Desktop/riscv_semi/target/riscv64imac-unknown-none-elf/release/deps/{libriscv_semihosting-9885f64fe96df006.rlib,libriscv_rt-51d68b33f01ce5fb.rlib,libriscv-59dab6363963a7ed.rlib,libcritical_section-e9f5889c719a3841.rlib,libriscv_pac-f89df8fa6230efac.rlib,libembedded_hal-5d1dcc1d7cb422dd.rlib,libpanic_halt-1ef63fdfdece3a56.rlib}" "/home/yash/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/riscv64imac-unknown-none-elf/lib/{librustc_std_workspace_core-7f86cd12f52bcad5.rlib,libcore-30a786714711bc78.rlib,libcompiler_builtins-40c1ef24ccedc865.rlib}" "-Bdynamic" "-z" "noexecstack" "-L" "/home/yash/Desktop/riscv_semi/target/riscv64imac-unknown-none-elf/release/build/riscv-rt-6ad9b6d4d4ba936a/out" "-o" "/home/yash/Desktop/riscv_semi/target/riscv64imac-unknown-none-elf/release/deps/riscv_semi-87ccd1d41771d2dd" "--gc-sections" "-O1" "--strip-debug" "-Tmemory.x" "-Tlink.x"
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: rust-lld: error: undefined symbol: _critical_section_1_0_acquire
          >>> referenced by riscv_semihosting.8936a401a212ccd8-cgu.0
          >>>               riscv_semihosting-9885f64fe96df006.riscv_semihosting.8936a401a212ccd8-cgu.0.rcgu.o:(riscv_semihosting::export::machine::hstdout_str::hfc1c719f4714c606) in archive /home/yash/Desktop/riscv_semi/target/riscv64imac-unknown-none-elf/release/deps/libriscv_semihosting-9885f64fe96df006.rlib
          
          rust-lld: error: undefined symbol: _critical_section_1_0_release
          >>> referenced by riscv_semihosting.8936a401a212ccd8-cgu.0
          >>>               riscv_semihosting-9885f64fe96df006.riscv_semihosting.8936a401a212ccd8-cgu.0.rcgu.o:(riscv_semihosting::export::machine::hstdout_str::hfc1c719f4714c606) in archive /home/yash/Desktop/riscv_semi/target/riscv64imac-unknown-none-elf/release/deps/libriscv_semihosting-9885f64fe96df006.rlib
          

error: could not compile `riscv_semi` (bin "riscv_semi") due to 1 previous error
yash@yash-VirtualBox:~/Desktop/riscv_semi$  ```

romancardenas, 2025-01-30:
That sounds more like you have not activated any feature flag to enable a valid implementation of `critical_section`. If your target is single HART, try enabling the `riscv`'s `critical-section-single-hart` feature.

romancardenas, 2025-01-30:
The patch is here: https://github.com/rust-embedded/riscv/tree/rvrt-patch

@yashwanthsinghm can you please run your apps using this branch and confirm it works?

imrank03, 2025-01-31:
> `riscv`'s `critical-section-single-hart` feature

Hi @romancardenas,

I made some changes and enabled the `riscv`'s `critical-section-single-hart` feature. However, I was still encountering an error. After applying your **patch** to update the dependencies, the code compiled without any errors. It seems there might be an issue with` riscv`, but I could be wrong.

Could you please check with `riscv` version `0.12.1`? Below is the relevant `Cargo.toml` configuration:

```Toml
[dependencies]
riscv = { git = "https://github.com/rust-embedded/riscv", branch = "rvrt-patch", features = ["critical-section-single-hart"] }
riscv-rt = { git = "https://github.com/rust-embedded/riscv", branch = "rvrt-patch" }
riscv-semihosting = { version = "0.1.2" }
panic-halt = "1.0.0"
```
Here is the working Git repository: (https://github.com/imrank03/riscv_semi).

Let me know your thoughts.

romancardenas, 2025-01-31:
You are experiencing an interesting issue regarding how Cargo deals with dependencies. Let's leave your `riscv` dependency pointing to `crates.io`.

The `riscv-rt` crate depends on `riscv`. However, when you use the GitHub version of `riscv-rt`, then `riscv-rt` depends on the GitHub version of `riscv`, not the one published in `crates.io`.

In your dependencies, you are mixing the `crates.io` version of `riscv` and the GitHub version of `riscv-rt`. As the GitHub version of `riscv-rt` depends on the GitHub version of `riscv`, you can see in the compilation traces that you are building two different versions of `riscv`:

```
    ...
   Compiling riscv v0.12.0 (https://github.com/rust-embedded/riscv?branch=rvrt-patch#f16c833d)
   Compiling riscv v0.12.1
   ...
```

Then, in your example, you don't use `riscv` at all. You only use `riscv-rt`, `riscv-semihosting`, and `panic-halt`. Thus, Cago just ignores your `riscv` dependency, which is not added. However, your `riscv` dependency is the one enabling the critical section feature!

Modify your example. You can add something like `use riscv as _;` (this is exactly the same trick used with `panic-halt`). Now, Cargo will notice that you want to use the `riscv` dependency in your Cargo.toml file and use it. I tested it in my machine and works.

Working examples:

```rust
#![no_std]
#![no_main]

use panic_halt as _;
use riscv as _; // We tell Cargo that we want to use our riscv dependency
use riscv_rt::entry;
use riscv_semihosting::{debug, hprintln};
// use semihosting::println;

#[entry]
fn main() -> ! {
    hprintln!("Hello, world!");
    debug::exit(debug::EXIT_FAILURE);
    loop {}
}
```

I prefer this other, though:

```rust
#![no_std]
#![no_main]

use panic_halt as _;
use riscv_rt::entry;
use riscv_semihosting::{debug, hprintln};
// use semihosting::println;

#[entry]
fn main() -> ! {
    hprintln!("Hello, world!");
    debug::exit(debug::EXIT_FAILURE);
    loop {
        riscv::asm::wfi(); // explicitly use riscv, so Cargo does not drop it
    }
}
```

yashwanthsinghm, 2025-02-01:
Thank you @romancardenas  for clearing out confusion, the patch branch works. I hope it reflects on crates.io in latest release.

romancardenas, 2025-02-18:
This should be fixed in the new release. @yashwanthsinghm can you check it?