rust-embedded/riscv #272: `riscv-rt`: example multicore.rs could cause HART to loop forever because of race condition

MLFlexer, 2025-02-28:
In the `multicore.rs` example, the other HARTs wait for an interrupt before entering `main`. However it seems as though there could be a race condition if HART 0 sets the interrupt flag to 1 in https://github.com/rust-embedded/riscv/blob/95cfb90c66fbfc0d66975a4235e1aeab96df9c39/riscv-rt/examples/multi_core.rs#L47 , and then HART 1 sets it to 0 in https://github.com/rust-embedded/riscv/blob/95cfb90c66fbfc0d66975a4235e1aeab96df9c39/riscv-rt/examples/multi_core.rs#L19.

This would make the interrupt flag be 0, when HART 1 is entering the loop in https://github.com/rust-embedded/riscv/blob/95cfb90c66fbfc0d66975a4235e1aeab96df9c39/riscv-rt/examples/multi_core.rs#L24-L29, and remain 0, because HART 0 has already set the flag, but it was overwritten by HART 1.

rmsyn, 2025-04-08:
If one wanted to introduce a global `Mutex` or `Atomic` (for lock-free algorithms), you could probably assure mutually-exclusive access to the portions that write to the `IPI` address.

However, as I understand the example, it is assuming this is a multicore platform, with a boot HART of `0`.

It sleeps the other cores until they receive an interrupt written to the `IPI` address.

All other HARTS would clear their `IPI` flag, and be stuck in the inner `wfi` loop until they receive another interrupt (breaking out if they receive an `msoft` interrupt).

So, if HART `0` sets the `IPI` for HART `1` before HART `1` enters `_mp_hook`, it would not break out of the inner loop because of the `wfi` call. Is that what you're saying?

However, aren't all of the HARTS going to be asleep before being woken by an `IPI` call? If you are assuming that HARTS were all woken by some previous software that hands control to this example, then I think you are correct, there is a potential race condition. And as mentioned above, you could use a `Mutex` or lock-free algorithm to control access to peripheral memory.

Feel free to make the example more robust against a potential race condition!

romancardenas, 2025-06-09:
In any case, the `_mp_hook` functions must be implemented in assembly, as it is executed before initializing RAM. Thus, we need to rework the example.