rust-embedded/riscv #295: Set _hart_stack_size = SIZEOF(.stack) for single-hart cores.

janderholm, 2025-06-07:
I got the following error while building for a riscv core.

```
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: rust-lld: error:
          ERROR(riscv-rt): .stack section is too small for allocating stacks for all the harts.
          Consider changing `_max_hart_id` or `_hart_stack_size`.

          rust-lld: error:
          ERROR(riscv-rt): .stack section is too small for allocating stacks for all the harts.
          Consider changing `_max_hart_id` or `_hart_stack_size`.
```

I can set _hart_stack_size to a smaller value, but I wonder if it would make more sense to set it to the size of the stack if the "single-hart" feature is enabled? They should effectively be the same anyway shouldn't they? This seem to work in my local memory.x:

`_hart_stack_size = SIZEOF(.stack);`

janderholm, 2025-06-07:
Should the ASSERT be changed to `>=` as well? Or am I off?

```ASSERT(SIZEOF(.stack) > (_max_hart_id + 1) * _hart_stack_size, "
ERROR(riscv-rt): .stack section is too small for allocating stacks for all the harts.
Consider changing `_max_hart_id` or `_hart_stack_size`.");
```

Currently if I set _hart_stack_size to 2048 and have a stack of 2048 this will result in a linking error.

romancardenas, 2025-06-09:
Hi @janderholm ! Thanks for opening the issue!

I think your proposals are reasonable. Would you like to start a PR? Otherwise, I can try to implement your idea.

Note that `_hart_stack_size = SIZEOF(.stack);` must only be active when the `single-hart` feature is on. To do feature-dependent changes in the `link.x`, we use the `build.rs` script with `${FEATURE}` wildcards.

janderholm, 2025-06-11:
PR posted!

romancardenas, 2025-06-19:
Closed as per #296

closed 2025-06-19 by romancardenas