rust-embedded/riscv #298: riscv-rt-macro: #[export_name = "main"] change to #[unsafe(export_name = "main")]?

xiaguangbo, 2025-06-12:
https://github.com/rust-embedded/riscv/blob/master/riscv-rt/macros/src/lib.rs#L139
why not have `edition = 2024`? add `edition = 2024` need change ` #[export_name = "main"]` to `#[unsafe(export_name = "main")]`

romancardenas, 2025-06-13:
Hi @xiaguangbo ! Thanks for raising this issue. Moving to Rust edition 2024 will need a few more tweaks, especially in macros that use `no_mangle` and `export_name`.  We can try to update the edition for all our crates :)

vm-kit, 2025-06-22:
When is enum macro used vs the single field?

```rs
#[cfg(not(target_arch = "riscv32"))]
read_write_csr_field! {
    Hstatus,
    vsxl: [32:33],
}
```

```rs
csr_field_enum! {
    VSBE {
        default: LittleEndian,
        LittleEndian = 0,
        BigEndian = 1,
    }
}
```

But i had seen a single bit flag used for enums i believe in another register implementation