rust-embedded/riscv #323: Riscv-rt not working at 80200000 with OpenSBI

vm-kit, 2025-07-02:
```rs
#![no_std]
#![no_main]

use core::arch::asm;
extern crate panic_halt;

#[riscv_rt::entry]
fn main() -> ! {
    unsafe {
        //let m = mstatus::read();
        asm!(
            "
            li s1, 0x10000000     // s1 = 0x1000_0000 i.e. our UART base address
            la s2, message2        // s2 = <str>
            addi s3, s2, 14       // s3 = s2 + 14, Calculating the address of the end of the string
            1:                    // Only labels of the form `<number>:`
                                  // does Rust allow when doing inline assembly.
            lb s4, 0(s2)          // s4 = (s2) i.e. Loading the byte at the memory
                                  // address stored in s2 into s4 
            sb s4, 0(s1)          // (s1) = s4 i.e. Store the byte from s4 to the 
                                  // memory address stored in s1
            addi s2, s2, 1        // s2 = s2 + 1
            blt s2, s3, 1b        // if s2 < s3, branch back to 1
            ",
            options(nostack)
        );
    }
    loop {}
}

#[unsafe(no_mangle)]
static message2: [u8; 14] = *b"Hello, enter!\n";
```
I'm using version 1.0 of panic_halt. 0.14 of riscv, and riscv-rt.

Here is my config.toml
```
[target.riscv64imac-unknown-none-elf]
runner = "riscv64-unknown-none-elf"
rustflags = [
  "-C", "link-arg=-Tmemory.x", # memory.x must appear BEFORE link.x
  "-C", "link-arg=-Tlink.x",
]

[build]
target = "riscv64imac-unknown-none-elf"
```
Here is my `memory.x` I have tried a few different values.

```
MEMORY
{
  RAM (rwxa) : ORIGIN = 0x80200000, LENGTH = 16M
}

REGION_ALIAS("REGION_TEXT", RAM);
REGION_ALIAS("REGION_RODATA", RAM);
REGION_ALIAS("REGION_DATA", RAM);
REGION_ALIAS("REGION_BSS", RAM);
REGION_ALIAS("REGION_HEAP", RAM);
REGION_ALIAS("REGION_STACK", RAM);

//_stext = ORIGIN(RAM);
//_heap_size = 0x1000;
//_max_hart_id = 1;
//_hart_stack_size = 0x1000;
//_stack_start = ORIGIN(RAM) + LENGTH(RAM);
```

If I run qemu with -bios none and the RAM address at 0x80000000. I can get the Hello World uart output. but if i run with `-bios default` giving me the open sbi output, I think the reset/trap is just going off. I do not see the uart output

vm-kit, 2025-07-02:
Sure. @romancardenas it's on gitlab, because I'm pretty over Microsoft stuff.

[gitlab](https://gitlab.com/vm-kit/vm-kit/-/tree/rt-version?ref_type=heads)

romancardenas, 2025-07-04:
Sorry for not responding, I'll try to take a look ASAP

vm-kit, 2025-07-04:
> Sorry for not responding, I'll try to take a look ASAP

@romancardenas no need to look into it now. I got that working branch running. But I'm not 100% got it working vs the nonworking one.

closed 2025-07-04 by vm-kit