rust-embedded/riscv #339: New releases

MabezDev, 2025-09-03:
Based on https://github.com/rust-embedded/riscv/milestone/1, it looks like https://github.com/rust-embedded/riscv/issues/239 is the only remaining issue. I believe we have all the required changes for us to switch back to using riscv-rt, and our PR is blocked on a new release: https://github.com/esp-rs/esp-hal/pull/3857.

Obviously we are not the only users of riscv-rt, but I'm wondering what's left for a release? I suppose https://github.com/rust-embedded/riscv/pull/299 would be good to get in, but after that do you think we can cut a release?

cc: @bjoernQ

romancardenas, 2025-09-03:
I was thinking of starting a release too!

Rust edition 2024 can wait. If I'm not wrong, Ferrocene does not support edition 2024 yet. As these crates are quite foundational, I think it is better to wait a bit before upgrading. I will keep the PR updated, so when the time comes, it will be easy to upgrade.

MabezDev, 2025-09-03:
Great! Anything we can help with, other than reviewing?

romancardenas, 2025-09-03:
As you wish. I cannot advance on this until 8 pm more or less. If you have some spare time you are more than welcome to update changelog files etc for the new release versions and I will review it later :)